PX4/DriverFramework #3: Test failing on Mac OS

LorenzMeier, 2015-11-06:
@mcharleb I get this output on Mac OS. Do you have specific pointers as how to debug?

```
Added driver 0x7fff538407d0 /dev/test0
TEST 1: readMessages
Read 3 messages
message 0: 9987
message 1: 9988
message 2: 9989
test PASSED (0)
TEST 2: read
Read 12 bytes (message = 4 bytes)
Read 3 messages
message 0: 20007
message 1: 20008
message 2: 20009
test PASSED (0)
TEST 3: ioctl
ioctl TEST_IOCTL_RESULT received
test PASSED (0)
TEST 4: Polling blocking read
Read 3 messages
message 0: 20010
message 1: 20008
message 2: 20009
Read 3 messages
message 0: 20010
message 1: 20008
message 2: 20009
test PASSED (0)
TEST 5: Polling timeout
Stopping driver and waiting for timeout
reading
timed out 60
test PASSED (60)
TEST 6: Read after setSampleInterval
start
reading
timed out 60
test FAILED (60)
tests done
shutdown
```

mcharleb, 2015-11-06:
In test/testdriver.hpp add a printf
```
virtual void _measure()
        {
                static int i = 0;
                m_lock.lock();
                m_message[i % m_count].val = i;
                i++;
                updateNotify();
                m_lock.unlock();
                printf("measure\n");
        }
```
if you don't see this printing out after "start" or "reading" then the WorkItem was not rescheduled.

LorenzMeier, 2015-11-06:
I'm not getting it:
```
TEST 6: Read after setSampleInterval
start
reading
timed out 60
test FAILED (60)
tests done
shutdown
```

mcharleb, 2015-11-06:
I'm fixing up pressure sensor unit test, then I'll try to debug it via Travis....

mcharleb, 2015-11-06:
@LorenzMeier In SyncObj::waitOnSignal(), the pthread_mutex_timedwait() is returning "60". What does than mean on MacOS? ETIMEDOUT? What is the value in errno.h?

int SyncObj::waitOnSignal(unsigned long timeout_ms)
{
        int ret;
        DEBUG("wait %p\n", &m_new_data_cond);
        if (timeout_ms) {
                struct timespec ts = absoluteTimeInFuture(timeout_ms);
                ret = pthread_cond_timedwait(&m_new_data_cond, &m_lock, &ts);
        }
        else {
                ret = pthread_cond_wait(&m_new_data_cond, &m_lock);
        }

        return ret;
}

mcharleb, 2015-11-07:
From the last build log on MacOS:
start
2209223 DevObj::start
2209225 DevObj::start 2
reading
timed out: Operation timed out (ret: 60)
test FAILED (60)
2209240 DevObj::removeHandle (0x7fff5aa1d170)
2209244 DevObj::stop

That means that the time from the read starting to the reported timeout (which should be 1 second) was actually 15 usec!

mcharleb, 2015-11-07:
2154793 TEST 6: Read after setSampleInterval
2154807 HRTWorkQueue::process In while
2154816 schedule
2154822 waiting for work (10000000)
2154832 schedule 2
2154838 HRTWorkQueue::process In while
2154853 schedule 3
2154876 waiting for work (10000000)
2154905 HRTWorkQueue::process In while
2154917 start
2154920 DevObj::start
2154922 DevObj::start 2
2154923 reading                                                         <<< test_read starts here with 1000ms timeout
2154926 HRTWorkQueue::process work exists
2154930 timed out: Operation timed out (ret: 60)       <<< operation timed out 7 usec later 
2154932 waiting for work (49978)
2154937 test FAILED (60)
2154952 HRTWorkQueue::process In while
2154965 DevObj::removeHandle (0x7fff5d1b8170)
2154972 HRTWorkQueue::process work exists
2154980 DevObj::stop
2232567 HRTWorkQueue::process do work (0x102a56a70)
2232630 DevObj::updateNotify
2232656 tests done
2232663 DevObj::stop
2232666 shutdown
2232667 HRTWorkQueue::process work exists
2232694 waiting for work (49925)
2232705 HRTWorkQueue::process In while
2232748 waiting for work (10000000)

mcharleb, 2015-11-07:
@LorenzMeier Fixed in https://github.com/PX4/DriverFramework/pull/4

PR #104 (2016-06-13): test: get unit tests on Mac working again
Mac doesn't seem to be behaving very accurately, however, we can make
the tests pass by being quite generous. The proper approach might be to
play with scheduling options on Mac OS.

Should fix #3.
closed 2015-11-07 by mcharleb