PX4/DriverFramework #20: Support for I2C ESCs

julianoes, 2016-02-10:
We should support ESCs over I2C.

http://www.rcgroups.com/forums/showthread.php?t=2032252

mhkabir, 2016-02-11:
I've been evaluating SimonK UART ESCs for use with the Snapdragon Flight. I believe it might be more robust compared to i2c.
I use these ESCs : http://www.hobbyking.com/hobbyking/store/__43709__Afro_ESC_20Amp_Multi_rotor_Motor_Speed_Controller_SimonK_Firmware_.html

They have exposed pads/pins under the shrinkwrap breaking out both the UART and the I2C. Uses a very simple UART protocol [https://github.com/sim-/tgy/blob/master/tgy.asm#L1737] and the Mikrokopter I2C protocol (which already has an older Device framework driver)

My Flight Kit arrives tomorrow, should be able to do some more concrete work on that then.

julianoes, 2016-02-11:
@LorenzMeier FYI

LorenzMeier, 2016-02-17:
I got these now at ETH and we'll move on with interfacing them.

julianoes, 2016-02-19:
I've been looking for info which pins/pads are for I2C but no success yet for the Afro 12A. If anyone knows, let me know.

mhkabir, 2016-02-19:
@julianoes Its on the silkscreen.
[image]

LorenzMeier, 2016-02-19:
Will be smaller on the 12 A. We also have 20A if needed.

mhkabir, 2016-02-19:
Checked my 12A's. The pad pinout is the same, but nothing printed on the silkscreen (no space, likely.) The 20A+ have the silkscreen print.

julianoes, 2016-02-19:
Ah, thanks @mhkabir, I was gonna say it's not on the silkscreen.

mhkabir, 2016-02-20:
@julianoes Quick (off topic) question -  Can the Snapdragon power module do 3S?

julianoes, 2016-02-22:
@mhkabir According to https://www.intrinsyc.com/snapdragon-embedded-development-kits/qualcomm-snapdragon-flight/ it says 2-6s.

mhkabir, 2016-02-22:
Ah cool, thanks!

julianoes, 2016-03-03:
I tried talking over i2C to the a stock Afro ESC 12A but it didn't respond yet.
I'll have to check with a scope next.

julianoes, 2016-03-17:
I don't have time for this right now, anyone interested feel free to pick it up.

mhkabir, 2016-03-20:
I'm taking this one up. I'll be using the UART, if that's not an issue.

mhkabir, 2016-03-20:
This is working now on a Pixhawk. Adding a i2c mode to my "simonk_esc" driver should be very trivial. I might even do it myself when I get some time.

@julianoes @LorenzMeier Can you please get me that debug adapter so that I can bring my Snapdragon board back up? Thanks.

TomPaynter, 2016-03-21:
The SimonK firmware partially supports the Mikrokopter I2C protocol. This allows the motor speed to be controlled by sending two-byte commands over I2C.  Data can be read back from the ESC over I2c, however in the standard SimonK firmware these values are constants that do not cause errors in Mikrokopter flight controller.

To drive more than one ESC, however, each ESC needs to be programmed with a different I2C address. We did this by modifying the SimonK firmware tgy.asm file and building a set of four .hex files; this can be done through the the AVRA compiler:

`avra ASM_FILE.asm`

If the ESC already has a SimonK firmware version since 2013, the resulting .hex can be easily programmed over the PWM signal line.  The USB programmer is http://www.hobbyking.com/hobbyking/store/uh_viewitem.asp?idproduct=39437

Using linux and the AVRDUDE programmer: 

`avrdude -p m8 -P PROGRAMMER_PATH -c avrispv2 -b 9600 -e -U flash:w:HEX_FILE.hex:i`

We have verified this works with new 20 A http://www.hobbyking.com/hobbyking/store/uh_viewitem.asp?idproduct=43709 and the 12 A Afro boards http://www.hobbyking.com/hobbyking/store/__55241__Afro_ESC_12Amp_BEC_UltraLite_Multirotor_ESC_V3_SimonK_Firmware_.html.  Note that on the smaller 12 A the I2C pins are not labeled, however their pinouts are identical to the 20 A board in the photo above.

julianoes, 2020-01-15:
So long :wave:.

closed 2020-01-15 by julianoes