PX4/DriverFramework #23: i2c write() leads to resets

julianoes, 2016-02-14:
@mcharleb, @jywilson
When the write of i2c.c is used, this leads to different ADSP restarts.
This needs investigation and fixing to make the HMC5883 driver work properly.

It is not clear to me if this bug is just because there is a leftover printf in i2c.c (see https://github.com/PX4/DriverFramework/issues/22), however, I've seen ADSP restarts just because of printfs, so I wouldn't be surprised.

julianoes, 2016-02-15:
@jywilson: to reproduce:

Checkout the latest [feature_mpu](https://github.com/PX4/DriverFramework/tree/feature_mpu) branch.

```
make
cd build_qurt
make df_mag_test-load && make libdf_mag_test-load
```

If it doesn't compile anymore, try ```git clean -d -f -x```, for me some leftover folders with auto-saved files made cmake stumble.

Run it:
```
cd /home/linaro
./df_mag_test_app
```

Without sensor attached, it will just hang, however you could probably try to comment out the while loop in [test.cpp](https://github.com/PX4/DriverFramework/blob/feature_mpu/drivers/bmp280/test/test.cpp) or only make it run once.

jywilson, 2016-02-19:
Hey Julian,

I checked the aDSP static image on my end and I don't see a crash, but I do see an I2C bus error.  I also don't see the extra debug output you mentioned.  Please try to obtain an updated static image ("Post CS Release"), to at least prevent the crash.  I am guessing, but I suspect that you see a crash because your older static image does not handle the I2C bus error correctly.

Regarding the bus error, the question remains how to fix this, and I have some reference for you that should help.  First, regarding the MPU9250 iteself, the magnetometer is actually a separate sensor that communicates over an internal auxiliary I2C bus.  To obtain access to this bus you need to enable "pass-through" mode.  Information on pass-through mode is available in the Invensense data sheet, available here: http://store.invensense.com/datasheets/invensense/MPU9250REV1.0.pdf

A good example of how to enable this mode in source code is available here: https://github.com/kriswiner/MPU-9250/blob/master/MPU9250BasicAHRS.ino.

Refer to the initMPU920() function for an example of how to enable pass-through (or bypass, as the code calls it) mode.  This code actually accesses each sensor by separate I2C bus transactions, treating the one sensor as three different sensors.  

Another option in this sensor, is to use the SPI bus to read the data from all three sensors in one bulk read.  This is the most efficient approach.  Since the existing PX4 MPU9x50 driver uses the SPI bus, perhaps you could extend this driver to also read the magnetometer data.  The steps required to intiialize the sensor for this mode of operation are listed below.  Contact me directly and I can provide more details.

   // Reset device

   // Enable stable clock source, auto select the best available clock source
   // PLL if ready, else use the internal 20MHz oscillator

   // Set 4KHz accel sample rate
   // MPU6500 shares 4kB of memory between the DMP and the FIFO. Since the
   // first 3kB are needed by the DMP, we'll use the last 1kB for the FIFO.

   // Turn on gyro and accelerometer

   // Disable all INTs

   // Disable FIFO on all sensors

   // Disable FIFO operaiton mode, Disable I2C Master I/F module and DMP

   // Disable I2C mode

   // Reset FIFO

   // Detect gyro

   // Initialize gyro with specified configurations

   // Detect and initialize compass using specified configurations

Jim W.

julianoes, 2016-02-19:
@jywilson Thanks for your pointers. Note that I'm currently trying to talk to an HMC5883 on the external I2C bus and not the mag which is internal in the MPU9250.

At this point, I'm blocked on this until I get a new static image.

julianoes, 2016-02-19:
Until I get the new static image and can further debug this, I need to workaround it by using continuous mode in the HMC5883 driver which does not require an I2C write in each measurement cycle, see: 07257d91c73e0540ffe3867cb6f52c4a9057bc4c.

julianoes, 2016-03-01:
@jywilson To reproduce this, you can use the branch:
https://github.com/PX4/DriverFramework/tree/debug_mag

julianoes, 2016-03-02:
This was due to a leftover printf, see #22. Fixed in the latest static aDSP image.

closed 2016-03-02 by julianoes