PX4/DriverFramework #24: Remove confusing errors/warnings

julianoes, 2016-02-14:
@mcharleb, @jywilson:

When using the DriverFramework or PX4 Firmware on the Snapdragon Flight, there are various confusing warnings/errors.

I would assume that they need fixing or removal. Otherwise, once more users start using the Snapdragon Flight, there will be plenty of confusion and support burden for us.

I'm referring to output like this:
```
/eagle-build/buildtree/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:136:failed to create tls key/eagle-build/buildtree/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:112:listener using ion heap: -1
/eagle-build/buildtree/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/apps_std_imp.c:35:apps_std fopen failed: /usr/share/data/adsp//dev/mag0 No such file or directory
/eagle-build/buildtree/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/inc/mod_table_imp.h:346::error: 2: 0 == (nErr = invoke_func_ptr(sc, pra))
```

julianoes, 2016-03-16:
The only warning remaining currently is:

```
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:136:failed to create tls key/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:112:listener using ion heap: -1
```

jywilson, 2016-05-05:
I will work on this getting rid of this warning.  May take a few days though, because this code change is in a driver I don't usually maintain.

julianoes, 2016-05-05:
It's definitely confusing "all" or at least most users in the beginning, so it would be nice to get rid of it.

julianoes, 2018-02-22:
Stale, closing.

closed 2018-02-22 by julianoes