PX4/DriverFramework #25: Compile error: sys/ttydefaults.h

julianoes, 2016-02-14:
dspal/include/termios.h:100:10: fatal error: 'sys/ttydefaults.h' file not found

Where would this file be? What is it for?

LorenzMeier, 2016-02-14:
Its a BSD standard header. Either missing or the QuRT default include path is missing.

julianoes, 2016-03-03:
A current workaround is pushed here: https://github.com/PX4/dspal/commit/eba0a18728aace328219d8985a08c7a506886b11

jywilson, 2016-03-07:
DSPAL does not currently support the termios API's.  The same functionality is provided in the form of IOCTL's.

julianoes, 2016-03-08:
@jywilson Understood, so the workaround (see https://github.com/PX4/dspal/commit/eba0a18728aace328219d8985a08c7a506886b11) is wrong?

julianoes, 2016-03-16:
Closing for now, until there is termios support, see #30.

closed 2016-03-16 by julianoes