PX4/DriverFramework #26: pollfd not defined in poll.h

julianoes, 2016-02-14:
It seems that ```struct pollfd``` is not available in poll.h when compiling for QURT.
Instead termios.h needs to be included.

See http://pubs.opengroup.org/onlinepubs/009695399/basedefs/poll.h.html

LorenzMeier, 2016-02-14:
Currently looks like this might require non-standard / non-POSIX compliant use of callbacks.

LorenzMeier, 2016-02-14:
https://github.com/ATLFlight/dspal/blob/master/test/dspal_tester/adsp_proc/serial_test_imp.c

julianoes, 2016-02-15:
https://github.com/ATLFlight/dspal/blob/master/test/dspal_tester/adsp_proc/serial_test_imp.c#L212

mcharleb, 2016-04-12:
@jywilson Is poll() supported in DSPAL?

jywilson, 2016-05-05:
No, poll is not supported in DSPAL.  It's something we can do but it would not be a trivial change.

Can you give me some idea of how important such a change is, just in terms of trying to decide what to work on first?

julianoes, 2016-05-05:
All the issues here labelled with "enhancement" are nice-to-haves but not blocking.

Of importance are internal mag and `/dev/mem` locking.

LorenzMeier, 2016-12-09:
@jywilson Is it feasible to remove poll() from your header for the time being then? It's super confusing that it does compile but doesn't work. What is the timeline to add it?

jywilson, 2016-12-10:
What headers are you referring to?  I couldn't find a reference to a poll function in the DSPAL headers, the Hexagon SDK headers, or the Hexagon Tools headers.

julianoes, 2018-02-22:
Stale, closing.

closed 2018-02-22 by julianoes