PX4/DriverFramework #30: Link to termios libs

julianoes, 2016-02-14:
How can we link to the termios libs?
Help would be appreciated, thanks.

```
[08500/03]  05:46.070  HAP:45:undefined PLT symbol fcntl (135) /libpx4muorb_skel.so  0303  symbol.c
[08500/03]  05:46.070  HAP:45:undefined PLT symbol tcgetattr (1316) /libpx4muorb_skel.so  0303  symbol.c
[08500/03]  05:46.070  HAP:45:undefined PLT symbol cfsetispeed (891) /libpx4muorb_skel.so  0303  symbol.c
[08500/03]  05:46.070  HAP:45:undefined PLT symbol cfsetospeed (1300) /libpx4muorb_skel.so  0303  symbol.c
[08500/03]  05:46.070  HAP:45:undefined PLT symbol tcsetattr (257) /libpx4muorb_skel.so  0303  symbol.c
```

julianoes, 2016-02-15:
@mcharleb:
> Add to LINK_LIBS in src/firmware/qurt/CMakeLists.txt

Add what? Adding termios didn't work.

mcharleb, 2016-09-23:
@jywilson Jim, can you chime in as to whether a new aDSP image is required and what version?

julianoes, 2016-10-05:
Support for termios coming in https://github.com/ATLFlight/dspal/pull/10.
Let's close this as soon as the next static image is out.

julianoes, 2018-02-22:
This has been in for a while, closing.

closed 2018-02-22 by julianoes