PX4/DriverFramework #33: open( ) uses m_dev_instance_path

bo-rc, 2016-02-19:
Why we need to use `m_dev_instance_path` in the [`open()` function](https://github.com/PX4/DriverFramework/blob/a11450bbadc599a1a4478aa5e7e60675a6c07275/framework/include/SPIDevObj.hpp#L93)? i have to change it to `m_dev_path` to make RPi2 find the sensor.

mcharleb, 2016-02-20:
Use DevMgr::getHandle(h) to open  a DriverFramework device. It will find the device instance if the devices is implemented as a real or virtual device.

bo-rc, 2016-02-21:
@mcharleb Thank you very much for commenting! What you said makes sense to me. However, there is a call in MPU9250.cpp: [This line](https://github.com/PX4/DriverFramework/blob/a11450bbadc599a1a4478aa5e7e60675a6c07275/drivers/mpu9250/MPU9250.cpp#L219) in `start()` which calls `DevOpen()`  to open a device without using a handle and it tries to open `m_dev_instance_path`. The correct path is in `m_dev_path` instead. This line fails to open the device. 

Should we correct it by modyfing MPU9250.cpp? For example, change the above-mentioned line to:
```cpp
int result = ::open(m_dev_path, 0);
```
will make the code pass all tests on the raspberry pi 2 system.

but the style looks a little bad. Any suggestions on this?

bo-rc, 2016-02-22:
After tracing the function calls, i think
the block of code starting with [`int result = devOpen(0);`](https://github.com/PX4/DriverFramework/blob/a11450bbadc599a1a4478aa5e7e60675a6c07275/drivers/mpu9250/MPU9250.cpp#L219) in `MPU9250::start()` might need to be deleted. Because it attempts to open the device `m_dev_instance_path` using an `open()` call. The following bock of code starting with [`result = SPIDevObj::start();`](https://github.com/PX4/DriverFramework/blob/a11450bbadc599a1a4478aa5e7e60675a6c07275/drivers/mpu9250/MPU9250.cpp#L225) aims the exact same thing and uses the correct path: `m_dev_path`.

Replacing those above-mentioned two code blocks with a unified [code block](https://github.com/bo-rc/DriverFramework/blob/11de3b904225cc467453623943f8657a66018f68/drivers/mpu9250/MPU9250.cpp#L220):
```cpp
	int result = SPIDevObj::start();
	if (result != 0) {
		DF_LOG_ERR("DevObj start failed");
		DF_LOG_ERR("Unable to open the device path: %s", m_dev_path);
		return result;
	}
```
solves the problem and all MPU9250 tests pass on the RPi2 system. See also this PR #32 .

@julianoes FYI

julianoes, 2016-03-17:
Fixed in #32/#42.

closed 2016-03-17 by julianoes