PX4/DriverFramework #44: Support for Navio2

julianoes, 2016-03-17:
We currently have support for Navio+ but not for Navio2 yet.

bo-rc, 2016-03-17:
Thank you @julianoes , i am getting a Navio2 so i will be able to work on both. I'm busy recently but i should be able to get back to development by this weekend.

julianoes, 2016-03-17:
Nice, looking forward to that!

blankered, 2016-03-29:
I have a Navio 2 and Navio +. Let me know if you need anything tested or how to help.

LorenzMeier, 2016-03-30:
@blankered Could you try to run the Navio 2 config on it?

nicolaerosia, 2016-04-18:
RPi3, compiled.
./build_rpi2/drivers/mpu9250/test/df_imu_test
2073442 count: 1000
2073453 IMU: accel: [-0.19, 2.71, 9.47] m/s^2
2073476      gyro:  [-0.03, -0.00, 0.00] rad/s
2073489      temp:  51.55 C
2073499 Closing IMU sensor

2073656 Test PASSED

LorenzMeier, 2016-04-18:
Nice! Would you be interested to push this further along until flight-readiness?

nicolaerosia, 2016-04-18:
I am interested, but I need a TODO list.

LorenzMeier, 2016-04-18:
Ok, so we already have an RPI config in the Firmware repo. Start off there, create a RPI3 config, then work through it to get all sensors up and running (the Snapdragon flight config should serve as starting point).

nacho-carnicero, 2016-04-28:
I'm also interested in contributing to the integration of PX4 on the RPi3/Navio2 configuration. I own the hardware so I can test code if needed, and I can contribute to the code although I'm not a big expert. Let me know if I can do something!

mayanez, 2016-06-06:
Hi,

I'm also interested in contributing support for RPi3/Navio2. Could someone post a TODO of the necessary drivers that are still missing?

Thanks!

julianoes, 2016-06-08:
@mayanez I don't actually know which drivers are working on which RPi1/2/3 and Navio1/2 hardware.

I'm going to create a todo list in this issue with the combinations and everyone can jump in and tick things off.

julianoes, 2016-06-08:
Strange enough, I just tried the `df_imu_test` on RPi2/Navio2 and it passed several times. I then rebooted and tried again, unfortunately, all I see now is:

```
18446744073709551608 Run it
706 SPI speed set to 1MHz.
1406 MPU9250 sensor WHOAMI wrong: 0x40, should be: 0x71
```

Reboots and power cycles don't seem to help.

staroselskii, 2016-06-08:
@julianoes 

I've been using DF's MPU9250 for a couple of days for testing Navio 2 and it seems to work fine always.
HMC5883 driver doesn't work, though. The reason is the lacking implementation of I2C drivers for Linux, of course.

staroselskii, 2016-06-08:
UPD:

I went on and on and eventually got it to the broken state. It looks different, though:

```
18446744073709551607 Run it
672 SPI speed set to 1MHz.
805 SPI speed set to 1MHz.
101040 Reset MPU9250
1233571 I2C transfer to mag timed out
1233678 MPU9250 soft reset failed.
1233720 mag initialization failed 1 tries
2366057 I2C transfer to mag timed out
2366125 MPU9250 soft reset failed.
2366163 mag initialization failed 2 tries
```

Our old examples don't work as well after that. Only a power cycle helps.

What are these obscure 18446744073709551608 and 18446744073709551607? Is there a chance of a stack overflow somewhere in the ```alloca()``` using code?

mayanez, 2016-06-09:
@staroselskii What does your .config file look like? Also, what exactly is missing from the I2C drivers to get the HMC5883 working?

staroselskii, 2016-06-10:
@mayanez 
Yeah, sure Navio doesn't have a HMC5883 on board. I just plugged one in the one of the spare I2C ports.

I2C driver problem is very DF-specific. I mean guys, take a look at this [line](https://github.com/PX4/DriverFramework/blob/master/framework/src/I2CDevObj.cpp#L135).

mhkabir, 2016-06-10:
Ah, we didn't implement i2c transactions for Linux userspace. @mayanez Would you mind taking a look at implementing this? Thanks!

mayanez, 2016-06-13:
@mhkabir Sure that shouldn't be a problem. However, I'm thinking of prioritizing the devices on the Navio2 first before I take a look at that. I'm working on porting the LSM9DS1 and the MS5611 since it looks like those are the main devices missing.

My Navio2 is the mail so once I finish the first pass through, would someone be able to test it in the meantime?

mayanez, 2016-06-23:
I'm trying to figure out how if I need to write a driver for the GPS. Would the current UBlox drivers work? I'm not sure how to test the GPS.

LorenzMeier, 2016-06-23:
It should work just fine on POSIX. Just add it to the build.

julianoes, 2016-06-23:
Awesome @mayanez, thanks for tackling these drivers. I'll review and cross-test tomorrow.

HidenoriKobayashi, 2016-06-23:
Hi,
It seems that U-blox GPS in Navio2 is on SPI. Does the current driver support SPI?
I needed to use ublox-spi-to-pty which is a utility supplied by Emlid to get gps start.

bkueng, 2016-06-24:
@HidenoriKobayashi We've only used it via serial so far. But the interface in linux seems to be the same, to talk to an SPI device. So you can try something like `gps start -d /dev/spidev1.0`
Communication with the gps is here: https://github.com/PX4/Firmware/blob/master/src/drivers/gps/gps.cpp#L335

julianoes, 2016-06-24:
@bkueng interesting. I might give this a try later.

julianoes, 2016-06-24:
@staroselskii: 

> I went on and on and eventually got it to the broken state. It looks different, though:
```
18446744073709551607 Run it
672 SPI speed set to 1MHz.
805 SPI speed set to 1MHz.
101040 Reset MPU9250
1233571 I2C transfer to mag timed out
1233678 MPU9250 soft reset failed.
1233720 mag initialization failed 1 tries
2366057 I2C transfer to mag timed out
2366125 MPU9250 soft reset failed.
2366163 mag initialization failed 2 tries
```
Our old examples don't work as well after that. Only a power cycle helps.

Thanks for reporting, I get that too. It's tracked in #89 and I really don't know how to debug it further.

mayanez, 2016-06-24:
It looks like there are some differences in the configuration messages and such for UBX protocol over SPI so it won't work straight out of the box. However, Emlid has a utility that converts SPI to PTY which then allows for reusing the current implementation. So just make sure to start that utility before PX4 and GPS works.

It looks like the last step remaining before being flight ready is adding RC support. I will be looking into that soon.

brettdaman, 2016-10-27:
Any updates for running px4 on navio2? I would like to try it.
I do not require GPS support.
What are the steps todo so?

LorenzMeier, 2016-10-27:
Check out http://dev.px4.io - there are complete instructions.

closed 2016-10-27 by LorenzMeier