PX4/DriverFramework #66: Latest Emlid RPi2 image breaks DF

mhkabir, 2016-04-30:
```
782161 overflow
782199 SPI speed set to 1MHz.
783143 SPI speed set to 1MHz.
783239 SPI speed set to 1MHz.
783332 SPI speed set to 8MHz instead of 10MHz.
783389 ERROR: SPI transfer bytes request over limit
```
Console is spammed with these messages. @bo-rc could you have a look please.

julianoes, 2016-05-03:
Thanks for reporting. It seems bad news that they don't support bigger buffers. I guess we'll have to do multiple bulk reads for Raspberry Pi but the overhead might make reading of the FIFO buffer quite slow.

staroselskii, 2016-05-11:
@julianoes @mhkabir 

Hello guys! I launched the df_imu_test yesterday and not surprisingly got the same result.

I'm pretty sure the buffer size is configurable. The ```spidev.bufsiz``` hasn't changed anything, though.

I suppose, the problem is somewhere in ```bcm2835-spi``` driver that is the default one now (it'd been ```bcm2708-spi```before).

I'd be glad to help but I'm on vacation right now, so I won't be able to test it until next Thursday.

bo-rc, 2016-05-11:
Hi developers, sorry i am having a very difficult time right now. I thought
i could dive into the our development but i was wrong; i need another solid
month before i can help here.

I am watching the progress though. Sorry i cannot contribute now but i
will.

You are the best!

mhkabir, 2016-05-13:
@staroselskii Thanks Georgii, We look forward to your comments.
@julianoes Georgii is from Emlid and will be helping us clear up this issue.

staroselskii, 2016-05-19:
@mhkabir @bo-rc @julianoes 

Hello, guys! I looked into the code today and have a couple of questions.

- Why do we have a limit that equals 21 byte [here](https://github.com/PX4/DriverFramework/blob/master/framework/src/SPIDevObj.cpp#L306)? Doesn't ```bulkRead()``` suppose to handle transactions with bigger buffer sizes? Once I got rid of this condition, the test was reporting totally fine results.
- I don't understand the conditions needed to pass the test in case of MPUs using FIFO, i.e. ```data.read_counter``` is updated in a [loop](https://github.com/PX4/DriverFramework/blob/master/drivers/mpu9250/MPU9250.cpp#L480). That's why the test fails with ```m_read_attempts != m_read_counter``` [here](https://github.com/PX4/DriverFramework/blob/master/drivers/mpu9250/test/test.cpp#L106) with ```m_read_counter``` being ~8 times greater. And it does make sense.

I sent a couple of PRs to address some of the issues I stated above. I'd be glad to discuss it more thoroughly as I might've missed something.

LorenzMeier, 2016-05-19:
@staroselskii The RPI2 has not seen the level of love and commitment it should have, so please operate under the assumption that if you find issues and you are convinced they are real and you have a fix that that's the way to go forward. Do you get by now proper inertial sensor readings with PX4 and in QGroundControl? This is what I would recommend to work against. If you show me plots (Tools/Widgets -> Analyse) you will see if it works satisfactory.

I'm pretty confident that this is pretty close and can be fixed relatively quickly and it would be fantastic if you could drive it.

julianoes, 2016-05-26:
Is this issue fixed in https://github.com/PX4/DriverFramework/pull/82?

julianoes, 2016-07-05:
Looks like it's fixed.

closed 2016-07-05 by julianoes