PX4/DriverFramework #68: The DF timer includes drift

LorenzMeier, 2016-05-03:
@mcharleb @jywilson If you ask the DF for a callback at a 4000 us interval you're getting 4000 us + scheduling overhead, not 4000 us on average, which is a problem for us. I will look with @bkueng into this, but if you have availability it would be great if you could propose a fix.

jywilson, 2016-05-03:
The work queues might need to be optimized.  Also, we can experiment with thread priorities to see if we can make it more deterministic.  

Is there a particular target latency you would like to see?  What level of jitter from the target 4000 us would be considered acceptable?  

If you could produce some standalone sample code that demonstrates the problem, this would really help in validating that a potential fix is making a difference.

mcharleb, 2016-05-04:
I can change it to expire at last expiry time + delta vs current time + delta

LorenzMeier, 2016-05-04:
We see 4200 - 6000 us instead of 4000. NuttX gives us 4 us jitter. I would love to see something closer to that order of magnitude.

mcharleb, 2016-05-04:
It looks like this was fixed in #71

PR #71 (2016-05-05): HRTWorkQueue: avoid scheduling drift
This should fix #68.

It would also be a good idea to check if there's too much work & scheduling falls behind.

@LorenzMeier can you test if this improves things?

btw `make run` shows some failed tests, but the same happens on master (not reported as error in travis).
closed 2016-05-05 by LorenzMeier