PX4/DriverFramework #101: Drouble with mpu9250 on intel Edison

AD-Aleksandrov, 2016-06-06:
MPU9250.
Guys, help please!
I take data several times but after that the driver is hung. 
And the driver is working random time before to hang.

julianoes, 2016-06-07:
Oh, this runs on Edison? :smile: What git hash are you using? Are you running the unit test? Can you put a `DF_LOG_INFO` in _measure() and check what happens?

AD-Aleksandrov, 2016-06-07:
I run the latest version. Yes, unit test. And after your commit, nothing changed: the driver is hung after a while. 
I tried to understand where is bug, there is an opinion the driver is hung in SPIDevObj::_bulkRead, but i am not sure. 
Thank you.

julianoes, 2016-06-07:
Well, you'll have to be sure exaclty where it hangs, otherwise it's hard to understand and fix :).

AD-Aleksandrov, 2016-06-08:
Hang in SPIDevObj::_bulkRead, calling ::ioctl(), line 322

julianoes, 2016-06-09:
So, if the [ioctl](https://github.com/PX4/DriverFramework/blob/master/framework/src/SPIDevObj.cpp#L322) hangs, can you debug further inside that call, so in the SPI library on the edison?

AD-Aleksandrov, 2016-06-10:
What do you think about failed test: offsetTime(), does depend on it something?

julianoes, 2016-06-10:
`offsetTime()` is used for the scheduling of the DriverFramework but not inside `ioctl()` calls which happen in the respective drivers such as the SPI/I2C libs on the platform.

julianoes, 2016-07-05:
@AD-Aleksandrov We would need additional information and help for this. I'll close it for now, feel free to re-open if you have a fix or more info.

julianoes, 2016-07-12:
Presumably addressed in #123.

closed 2016-07-18 by AD-Aleksandrov