PX4/DriverFramework #119: i2c write fail on start

julianoes, 2016-07-05:
Latest master fails on PX4 startup:

```
1926824: HAP:61:In dspal_entry (main.cpp:214)
   1927100: HAP:61:attempting to open the ADSP command file: /dev/fs/px4.config (main.cpp:178)
   1927101: HAP:61:reading commands from /dev/fs/px4.config (main.cpp:185)
   1927156: HAP:61:Sleeping for 1, 1000000 (apps.h:118)
   1928102: HAP:61:1598159 Reset MPU9250 (MPU9250.cpp:89)
   1928154: HAP:60:1662432 IMU temperature initialized to: 55.210526 (MPU9250.cpp:496)
   1928157: HAP:61:1666599 additional sensor configuration succeeded (BMP280.cpp:187)
   1928158: HAP:61:start (df_trone_wrapper.cpp:203)
   1928160: HAP:61:I2C_Write failed: 285212872 num_bytes_written 0 (i2c.c:414)
   1928160: HAP:61:1669634 Error: i2c write failed. Reported -1 bytes written (I2CDevObj.cpp:215)
   1928160: HAP:61:I2C_Write failed: 285212872 num_bytes_written 0 (i2c.c:414)
   1928160: HAP:61:1669844 Error: i2c write failed. Reported -1 bytes written (I2CDevObj.cpp:215)
   1928160: HAP:61:I2C_Write failed: 285212872 num_bytes_written 0 (i2c.c:414)
   1928160: HAP:61:1670059 Error: i2c write failed. Reported -1 bytes written (I2CDevObj.cpp:215)
   1928160: HAP:61:I2C_Write failed: 285212872 num_bytes_written 0 (i2c.c:414)
   1928160: HAP:61:1670272 Error: i2c write failed. Reported -1 bytes written (I2CDevObj.cpp:215)
...
```

julianoes, 2016-07-05:
The driver causing the error messages seems to be `df_trone_wrapper`.

julianoes, 2016-07-05:
Fixed in #120.

closed 2016-07-05 by julianoes

stu85010, 2016-08-11:
Hi @julianoes , I thought the retries [method](https://github.com/PX4/Firmware/blob/master/src/drivers/device/i2c_nuttx.cpp#L191-L230) for Nuttx might prevent a potential overflow.
Please refer to [76507b6](https://github.com/stu85010/DriverFramework/commit/76507b6516475c90d2b918b6daf805e85155242c
).

julianoes, 2016-08-11:
Hi @stu85010, it looks like your fix is safer. Can you make a PR?

PR #120 (2016-07-05): Fix I2C retries
This fixes #119. The `_retries` variable was uninitialized which lead to many retries if unlucky.