PX4/DriverFramework #122: Fix collect logic in MS5607

LorenzMeier, 2016-07-06:
@eyeam3 This fix went in for the MS5611.

https://github.com/PX4/DriverFramework/commit/bc07efc58c40b0a66f80e4c0eee5d0cdaac2eeca

Are you 100% sure you need a different driver for the MS5607? How do they differ? I would prefer to have this merged into a single driver. We're already in maintenance hell.

julianoes, 2016-07-06:
I agree but I think we will have to do some refactoring first. We'll have to talk through the architectur on a whiteboard I'd suggest.

eyeam3, 2016-07-06:
I totally agree, that there is no need for two separate drivers. As far as I know, they only differ in `convertPressure` and `convertTemperature`.
I first wrote the driver for the MS5607, which was not tested in combination with other sensors and had that severe usleep-mistake. Sorry for that. The code was then copied to the MS5611.

LorenzMeier, 2016-07-06:
Would you have the chance to stop by in the lab at 3 or at 1?

eyeam3, 2016-07-06:
I am not in Zurich today, but I could stop by tomorrow, if this would be possible as well.

LorenzMeier, 2016-07-06:
Let's carry on the discussion via Github then. Julian and I will quickly look at it and recommend a design pattern.

eyeam3, 2016-10-04:
Has been fixed with #142.

closed 2016-10-04 by eyeam3