PX4/DriverFramework #133: Accelerometer readings are offset on some MPU9250 devices

ericye16, 2016-07-25:
On some MPU9250 devices, the accelerometer readings are offset. For example, this is the output of df_imu_test on a board that is sitting flat and stationary:
```
[08500/00]  01:18.777  HAP:12349:18446744073709551552 WARNING: setRealtimeSched failed (not run as root?)   0357  DriverFramework.cpp
[08500/00]  01:18.878  HAP:12350:100810 Reset MPU9250   0170  MPU9250.cpp
[08500/00]  01:18.885  HAP:12350:107897 gyro self-test: 0xbd, 0xcb, 0xe2   0061  MPU9250.cpp
[08500/00]  01:18.885  HAP:12350:108120 accel self-test: 0x6a, 0x64, 0x79   0070  MPU9250.cpp
[08500/00]  01:18.888  HAP:12350:110672 accel self-test readings: 0xdb94, 0x16a, 0xf93d   0099  MPU9250.cpp
[08500/00]  01:18.890  HAP:12350:113111 accel non-self-test readings: 0xdd06, 0x438, 0xfc8c   0115  MPU9250.cpp
[08500/00]  01:18.890  HAP:12350:113206 accel st and nonst diff readings: 0xfe8e, 0xfd32, 0xfcb1   0120  MPU9250.cpp
[08500/00]  01:18.890  HAP:12350:113300 accel st and nonst diff readings: 0x172, 0x2ce, 0x34f   0124  MPU9250.cpp
[08500/00]  01:18.953  HAP:12349:176300 IMU temperature initialized to: 47.542938   0587  MPU9250.cpp
[08500/00]  01:18.953  HAP:12350:176458 count: 1   0095  test.cpp
[08500/00]  01:18.955  HAP:12350:176517 IMU: accel: [-45.09, 1.06, -9.36] m/s^2   0143  ImuSensor.hpp
[08500/00]  01:18.955  HAP:12350:176709      gyro:  [-0.05, 0.00, 0.00] rad/s   0147  ImuSensor.hpp
[08500/00]  01:18.955  HAP:12350:176872      mag:  [0.000000, 0.000000, -0.000000] ga   0153  ImuSensor.hpp
[08500/00]  01:18.955  HAP:12350:176954      temp:  47.54 C   0157  ImuSensor.hpp
[08500/00]  01:18.955  HAP:12350:177152 count: 10   0095  test.cpp
[08500/00]  01:18.955  HAP:12350:177212 IMU: accel: [-45.11, 1.06, -9.19] m/s^2   0143  ImuSensor.hpp
[08500/00]  01:18.955  HAP:12350:177386      gyro:  [-0.05, 0.00, -0.00] rad/s   0147  ImuSensor.hpp
[08500/00]  01:18.955  HAP:12350:177530      mag:  [0.460793, -0.144281, 0.382002] ga   0153  ImuSensor.hpp
[08500/00]  01:18.955  HAP:12350:177693      temp:  47.45 C   0157  ImuSensor.hpp
[08500/00]  01:18.955  HAP:12350:177812 Closing IMU sensor   0119  test.cpp
[08500/00]  01:18.965  HAP:12350:188241 Test PASSED   0140  test.cpp
```

I don't think this is a hardware issue because when I move it, the numbers change but they are offset. I tried setting the offset parameter in the hardware but it seems to reset every boot. This wasn't happening before so is it possible a bad calibration or something persistently changes the offset?
I also tried looking into the self-test features on the IMU but invensense doesn't seem to be willing to release those docs :(

julianoes, 2016-07-25:
Hm, that looks very wrong. Are the axes correct? Is x still x, y still y, etc.?
Is this something that came up with a newer version or just a new board? If it came up through newer software you could do a `git bisect` to find the bad change.

I would argue it's a hardware fault because I have heard from at least 2 faulty MPU9250 on the Snappy boards that showed faults sort of similar to this, as in they had some values completely wrong and others ok.

ericye16, 2016-07-25:
Yeah, the axes are correct. It's weird because the board exhibiting this issue:
- did not exhibit this issue before
- when reverting to software releases from earlier, still doesn't work.

I found and ported over the MPU9250 self-test functionality from: https://github.com/kriswiner/MPU-9250

and I found that for the board that doesn't work, the accelerometer reading under test and not under test didn't differ (which shouldn't happen).
Self-test branch is here: https://github.com/ericye16/DriverFramework/tree/mpu9250_self_test

julianoes, 2016-07-26:
Ok, thanks for the info and link. Does that mean that the self test fails? And is there any other driver (version) out there that shows correct values?

ericye16, 2016-07-26:
Heh, turns out there was a bug in the code yesterday.

Today's results on faulty board:
```
[08500/00]  01:43.436  HAP:28734:100799 Reset MPU9250   0178  MPU9250.cpp
[08500/00]  01:43.645  HAP:28734:308797 selfTest[0]: 106   0113  MPU9250.cpp
[08500/00]  01:43.645  HAP:28734:310468 aSTAvg[0] = -19681   0131  MPU9250.cpp
[08500/00]  01:43.645  HAP:28734:310538 aAvg[0] = -23576   0132  MPU9250.cpp
[08500/00]  01:43.646  HAP:28734:310601 factoryTrim[0] = 7448.100586   0133  MPU9250.cpp
[08500/00]  01:43.646  HAP:28734:310729 aSTAvg[1] = 7017   0131  MPU9250.cpp
[08500/00]  01:43.646  HAP:28734:310793 aAvg[1] = -20   0132  MPU9250.cpp
[08500/00]  01:43.646  HAP:28734:310853 factoryTrim[1] = 7016.447754   0133  MPU9250.cpp
[08500/00]  01:43.646  HAP:28734:310948 aSTAvg[2] = -7098   0131  MPU9250.cpp
[08500/00]  01:43.646  HAP:28734:311012 aAvg[2] = -15597   0132  MPU9250.cpp
[08500/00]  01:43.646  HAP:28734:311075 factoryTrim[2] = 8647.013672   0133  MPU9250.cpp
[08500/00]  01:43.646  HAP:28734:311163 deviation is: -47.704784, 0.292921, -1.711732, -0.068182, 0.147069, 0.534487   0265  MPU9250.cpp
```

On a good board:
```
[08500/00]  01:34.695  HAP:63:305313 selfTest[0]: 102   0113  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:306954 aSTAvg[0] = 7497   0131  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:307018 aAvg[0] = 376   0132  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:307075 factoryTrim[0] = 7157.478516   0133  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:307198 aSTAvg[1] = 6660   0131  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:307258 aAvg[1] = -187   0132  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:307315 factoryTrim[1] = 6810.095215   0133  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:307403 aSTAvg[2] = -9665   0131  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:307463 aAvg[2] = -17160   0132  MPU9250.cpp
[08500/00]  01:34.697  HAP:63:307521 factoryTrim[2] = 7597.807617   0133  MPU9250.cpp
[08500/00]  01:34.698  HAP:63:307609 deviation is: -0.509655, 0.541914, -1.353121, -1.877409, -1.496136, 0.223827   0265  MPU9250.cpp
```

So it seems like the deviation of ~-40% might be notable...

julianoes, 2016-07-26:
I don't understand, so it's a hardware or software problem?

ericye16, 2016-07-26:
I'm not sure...it'd be nice if I could run this on other "failing boards." It might be a common hardware problem that could be worked-around in software but currently isn't. (MPU9250 does have offset registers, but it seems like we don't use them right now...)

I'll keep looking into it and update this as necessary.

crossa, 2017-03-13:
I have fixed it . change spi frequency to  5M not 20M
Because  mpu9250 worked as slaver and using CLK of mast as clock.
The max speed should be 20/4 = 5M not 20M

julianoes, 2017-03-13:
@crossa so this was not a hardware issue after all?

mhkabir, 2017-03-13:
@crossa Isn't running a Snapdragon Flight board incidentally. Maybe you can test on one of the faulty ones in the lab @julianoes ?

julianoes, 2017-03-13:
@mhkabir Oh yes I got that wrong. Need to check if I can find one.

crossa, 2017-03-14:
@Julianoes 

It's not hardware fault.

If using any mpu9250 as spi mode, please change the frequency.

1/4 of maxmum MCU frequency is allowed.

If using maximum frequency of MCU  as the spi speed and other sensor driver is running ,the data will be offset.

Beacause the data piling up faster than the master（host） can process it.

This trouble has been bothering me for over a month.This problem can casuse the copter crash apprehensive.

 I have read a lot of information to know how to solve this problem.

julianoes, 2017-03-14:
@crossa got it. Thanks.

julianoes, 2019-10-10:
This is probably outdated, closing.

closed 2019-10-10 by julianoes