PX4/DriverFramework #144: Macro inconsistencies

bkueng, 2016-09-19:
40ec8612c627fb15101b2 changed the macro `__DF_LINUX` to `__LINUX`. Bebop uses `__BEBOP`. Darwin uses `__DF_DARWIN`, and for nuttx there are multiple, even some `__PX4_NUTTX`.

So what should the convention be? @julianoes 

Also: the macro should be defined in `cmake/toolchains/`, but within PX4 Firmware, these never get included, and thus these macros need to be defined in the Firmware as well, which is prone to error. Case in point: https://github.com/PX4/Firmware/blob/master/cmake/posix/px4_impl_posix.cmake#L201, which should now be `__LINUX`.

mcharleb, 2016-09-23:
I don't think the PX4 NuttX build actually supoprt DriverFramework. I chose __DF_LINUX so that it would not conflict with any other  upper layer project uses of __LINUX.  Projects that use DF should set the appropriate flags (__DF_LINUX, __DF_DARWIN,...). I hadn't been following the additions of __BEBOP and the change to __LINUX. There are uses for DF outside PX4 so the flags should not be changed from __DF_LINUX to __LINUX or any other such change.  I agree they are currently inconsistent so I guess the choices are to use:

__LINUX, __DARWIN, __NUTTX, __BEBOP, __QURT, etc., or __DF_LINUX, __DF_DARWIN, __DF_NUTTX, __DF_BEBOP, __DF_QURT, etc.

bkueng, 2016-09-23:
Thanks for the clarifications.

> I chose __DF_LINUX so that it would not conflict with any other upper layer project uses of __LINUX

I agree and for the same reason I think we should stick to the `__DF_` prefix. It also makes it clear in the upper project where the macro belongs to.

> Projects that use DF should set the appropriate flags (__DF_LINUX, __DF_DARWIN,...).

That's something that should be documented then (is it already?). Maybe a README section 'How to include DF in your project' would be helpful here.

julianoes, 2016-09-27:
> I agree and for the same reason I think we should stick to the __DF_ prefix. It also makes it clear in the upper project where the macro belongs to.

Ok, I agree.

julianoes, 2020-01-15:
Most drivers have been moved to the PX4 Firmware repo, see https://github.com/PX4/Firmware/pull/13900.

I'm therefore closing the issues here.

closed 2020-01-15 by julianoes