PX4/DriverFramework #179: Sonar inoperative on Bebop 2

mhkabir, 2017-02-02:
@eyeam3 Using current PX4 master, my Bebop 2 sonar does not return any measurements, staying stuck at -1.0m. I get very rare spikes with the seemingly correct distance sometimes. I checked using QGC and `listener distance_sensor`.

Let me know what you need to debug this, or let me know how I can help.

eyeam3, 2017-02-02:
Can you activate logging on debug level in the DF driver and tell me what it prints? That would be really helpful.

mhkabir, 2017-02-02:
@eyeam3 How do I do that? Quick search didn't reveal any docs.

eyeam3, 2017-02-02:
There is a check for `DF_DEBUG` in the logging header. You can add `#define DF_DEBUG 1` in this line https://github.com/PX4/DriverFramework/blob/master/drivers/bebop_rangefinder/BebopRangeFinder.cpp#L35. Then `DF_LOG_DEBUG` will be printed.

mhkabir, 2017-02-02:
@eyeam3 Okay, it continiously spits out
```
46305203 Bellow block distance
...
```
even over a hard (marble) floor, at approximately 1m from the floor. On my bed, it performs slightly better and atleast returns readings, but it still spikes a lot, and goes to "`Bellow block distance`" often.

eyeam3, 2017-02-02:
Thanks, that helps a lot. I will check it on the weekend.

mhkabir, 2017-02-02:
Thanks! Let me know in case you need any information. I will try to debug it on my side tomorrow morning. Any pointers as to what I can try?

eyeam3, 2017-02-02:
You can increase this value: https://github.com/PX4/DriverFramework/blob/master/drivers/bebop_rangefinder/BebopRangeFinder.cpp#L52.

However, this is not a real fix, it is just an adjustment to your environment / ground. The implemented algorithm is very simple and searches for the rising edge of the reflected signal, but apparently this is not a robust solution. I will most likely replace the entire algorithm with something more sophisticated.

julianoes, 2020-01-15:
Most drivers have been moved to the PX4 Firmware repo, see https://github.com/PX4/Firmware/pull/13900.

I'm therefore closing the issues here.

closed 2020-01-15 by julianoes