PX4/DriverFramework #186: sample_interval_usec is not  essential for pca9685

crossa, 2017-02-24:
Now  I‘m tring to port  driver of  PCA9685 i2c module into DriverFramework.
I wrote a  Class named PCA9685 extends I2CDevObj and I found the last parameter of  I2CDevObj constructor named sample_interval_usec. This parameter is important for sensor drivers but is not necessary for pwm output.

I had read navio_sysfs_pwm_out.cpp and I found this driver run the main task without  interval and output pwm signal in a while loop.

So the question is :
Is the sample_interval_usec essential parameter for the construnctor of I2CDevObj？
Pwm output processing should be written in a loop and this loop should always run until the driver stopped, so how do I do？

bkueng, 2017-02-24:
if `sample_interval_usec` is 0, the periodic update method will not be called. If you need your own loop and scheduling, then the driver framework is probably not the correct place (or at least not in its current form). It has been designed for drivers with periodic, regular updates, running on a single worker thread. You might find drivers that do it differently within the DF, but I'm not aware of any.

crossa, 2017-02-24:
Ok. Thanks .
I have wrote a MPU9250 driver(work with raspberry pi on Linux, not nuttx) for px4 but it in the driver not driverframework. I thought  the df was for posix system，then I want to port my driver into the framework.
Since it's different,  so give it up

closed 2017-02-24 by crossa