PX4/DriverFramework #207: MPU9250 mag start failure

crossa, 2017-06-19:
When using  the latest version of DriverFramework
When start mpu9250 with mag,
"ERROR [sensors] Mag #0 fail:  STALE!"  dispayed in the pxh terminal and mpu9250 didn't work

Please check it

julianoes, 2017-06-19:
@crossa on which hardware?

crossa, 2017-06-19:
@julianoes  px4firmware v1.6.3 on linux based hardware such as raspberry pi ,nano pi.
I meet this issue yesterday.  I was going to do the upgrade operation but give it up .
I have tried find the cause of the error but faild.

crossa, 2017-06-20:
@julianoes  I think the px4firmware cause the iusse.  If "ERROR [sensors] Mag #0 fail: STALE!" display,
I  can calibrate the mpu9250 senor with mag in the QGC，but it's useless .After the calibration, the driver does not work. I use the mag of mpu9250 as mag #0, I don't know whether it will work correctly if I use another independent mag .

@LorenzMeier  please check it

crossa, 2017-06-20:
@julianoes @LorenzMeier 
At Firmware/src/modules/sensors/voted_sensors_update.cpp line 897
bool VotedSensorsUpdate::check_failover(SensorData &sensor, const char *sensor_name)
AND line 912.

I don't know why the mag can not pass the check

mhkabir, 2017-06-20:
@crossa It is fixed on master. Please retest. 
The fix has not been merged to stable yet : https://github.com/PX4/Firmware/pull/7434

crossa, 2017-06-20:
@mhkabir 
thank u . I'll retest it tonight

closed 2017-06-26 by crossa