PX4/DriverFramework #230: Priority way too low

bartslinger, 2018-09-28:
I am experiencing issues on Bebop 2 because the DriverFramework priority is hard-coded to **10** (!!). Basically everything else is running at a higher priority. This is terrible because now we have IMU and MAG on the lowest priority.

https://github.com/PX4/DriverFramework/blob/master/framework/src/DriverFramework.cpp#L333

However I'm not sure what would be an appropriate value here. That's what I'd like to discuss in this issue.

julianoes, 2018-09-29:
Interesting find! Have you tried changing it and tested the results?

bartslinger, 2018-09-29:
I got a ton of warnings like Accel TIMEOUT when running heavy calculations (opencv stuff) in another thread at prio 35. Then I changed DriverFramework priority from 10 to 97 and those warnings disappeared. On linux systems such as bebop, the priority range for SCHED_FIFO is 1 to 99. I don't know if other posix px4 systems exist with different priority ranges.

`wkr_hrt` and `bebop_bus_esc_main` are at prio 99. `hpwork` is at prio 98. There might be more because I only checked the tasks threads.

mhkabir, 2018-09-29:
@bartslinger Tangential question, are you running OpenCV onboard the Bebop?

julianoes, 2018-09-30:
@bartslinger to me it seems like it makes sense to bump the priority then.